brick-a-brack/eagle-animation #36: Linux Contributor Support

Swivelgames, 2023-04-20:
Seems there's an issue with lightningcss finding the correct node extension when building on Linux.

```python
@parcel/transformer-css: Could not resolve module "~/eagle-animation/node_modules/lightningcss/lightningcss.linux-x64-gnu.node" from "~/eagle-animation/node_modules/lightningcss/node/index.js"
Error: Could not resolve module "~/eagle-animation/node_modules/lightningcss/lightningcss.linux-x64-gnu.node" from "~/eagle-animation/node_modules/lightningcss/node/index.js"
    at $0578d0f6e116167e$export$fb2a0b866a8162dc.resolve (~/eagle-animation/node_modules/@parcel/package-manager/lib/index.js:4991:21)
    at NodePackageManager.resolveSync (~/eagle-animation/node_modules/@parcel/package-manager/lib/index.js:3510:42)
    at NodePackageManager.requireSync (~/eagle-animation/node_modules/@parcel/package-manager/lib/index.js:3352:34)
    at Module.m.require (~/eagle-animation/node_modules/@parcel/package-manager/lib/index.js:3366:25)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (~/eagle-animation/node_modules/lightningcss/node/index.js:21:22)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at NodePackageManager.load (~/eagle-animation/node_modules/@parcel/package-manager/lib/index.js:3375:15)
```

For those getting a similar error, the workaround seems to be to install the appropriate `lightningcss` extension manually:

In my case, it's looking for `lightningcss.linux-x64-gnu.node`, so the following resolved the error:

```ruby
npm i lightningcss-linux-x64-gnu --force
```

Don't forget the `--force` argument, since it's required to resolve other dependency issues in this project.

Good luck!

Maxou44, 2023-04-24:
Hello and thanks for the feedback 😀

Is ```npm i -D lightningcss --force``` works to fix this issue instead of a OS specific dependency?

Swivelgames, 2023-05-01:
I wholly agree. OS-specific dependencies should be avoided whenever possible. Adding `lightningcss` as a dependency directly appears to solve the issue.

Maxou44, 2023-05-01:
Thanks for the feedback, it will be added as dependency in the next update✌️

Maxou44, 2023-05-02:
Fixed on dev branch in commit 8b9f693, I close the issue ✌️

closed 2023-05-02 by Maxou44