brick-a-brack/eagle-animation #37: EagleAnimation should respect XDG user-dirs config

Swivelgames, 2023-05-01:
## Yikes!

Currently, EagleAnimation commits a very big no-no by not respecting XDG user-dirs config, which contributes to homedir clutter.

https://github.com/brick-a-brack/eagle-animation/blob/7edf4bb4607b94f6b931caa4c99198f7de784fc8/src/electron/actions.js#L14-L16

To put homedir clutter into perspective, I've recently spent a fair bit of time cleaning my homedir this evening. Here's what it looks like right now:
[image]

As you can see, despite the specification being widely adopted, there are _many_ applications that don't respect these directories. Heck, even Windows added the AppData directory in order to attempt to remedy this for Windows users many years ago, and many still don't.

## Drop-in Cross-platform Solutions!

However, there are solutions out there that allow you do so in a cross-platform manner, like [env-paths](https://www.npmjs.com/package/env-paths) or [@folder/xdg](https://www.npmjs.com/package/@folder/xdg), among others.

Without having to arduously peruse the [XDG Base Directory Specification], I believe the directory most appropriate for the type of content that's stored here is in `$XDG_DATA_HOME`. This puts all the data in a predictable location, without polluting a user's homedir.

You can reference this directory using `env-paths` with:

```javascript
import envPaths from 'env-paths';

// ...

const PROJECTS_PATH = envPaths(DIRECTORY_NAME).data;
```

Or, with `@folder/xdg`:

```javascript
import xdg from '@folder/xdg';

// ...

const PROJECTS_PATH = join(xdg().data, DIRECTORY_NAME);
```

This should resolve to the correct directory on all platforms:

- Linux: `$HOME/.local/share/EagleAnimation`
- macOS: `$HOME/Library/Application Support/EagleAnimation`
- Windows: `%APPDATA%\EagleAnimation`

## What about existing users?

In order to not affect current users of the application, you could simply check to see if `$HOME/EagleAnimation` exists and use that, otherwise use the appropriate path:

```javascript
import fs from "fs";
import envPaths from 'env-paths';

const OLD_PROJECTS_PATH = join(homedir(), DIRECTORY_NAME);
if (fs.existsSync(OLD_PROJECTS_PATH)) {
    const PROJECTS_PATH = OLD_PROJECTS_PATH;
} else {
    const PROJECTS_PATH = envPaths(DIRECTORY_NAME).data;
}
```
or
```javascript
const OLD_PROJECTS_PATH = join(homedir(), DIRECTORY_NAME);
const PROJECTS_PATH = fs.existsSync(OLD_PROJECTS_PATH) ? OLD_PROJECTS_PATH : envPaths(DIRECTORY_NAME).data;
```

## Going above and beyond!

Another option, of course, would be to _move_ the directory if it exists, but that can be dangerous. That would be more involved, because you would probably want to prompt the user when the application starts to see if they _want_ you to move it with a `yes`/`remind me later`/`no` prompt.

Maxou44, 2023-05-02:
I tried to work on it today, I have an issue because the 'env-paths' package is an ESM-only module and can't be imported, we also have this issue with the node-fetch package

Related to : https://github.com/electron/electron/issues/21457

Swivelgames, 2023-05-03:
Oh wow! That's pretty surprising. I would have expected `electron` to be one of the earlier projects to support ESM.

Maxou44, 2023-05-10:
Implemented on dev branch using v2.x.x version of the package

commit eade57a1 (2023-05-10): Use XDG user-dirs, fixes #37
closed 2023-05-10 by Maxou44