brick-a-brack/eagle-animation #63: DRAGONFRAME BLUETOOTH/USB CONTROLLER

Reaper10, 2024-04-03:
add support for  DRAGONFRAME BLUETOOTH/USB CONTROLLER
https://www.dragonframe.com/product/dragonframe-bluetooth-controller/
you may need to function

Maxou44, 2024-04-04:
Hello @Reaper10 and thanks for your issue 😊

I don't have a Dragon Frame controller to test it but it should work as it is detected as numpad, 😉

Can you try and give us feedback about what works and what doesn't? I will fix it 👌

Reaper10, 2024-04-04:
> Hello @Reaper10 and thanks for your issue 😊
> 
> I don't have a Dragon Frame controller to test it but it should work as it is detected as numpad, 😉
> 
> Can you try and give us feedback about what works and what doesn't? I will fix it 👌

i dont have either thout it would goof