brick-a-brack/eagle-animation #67: Webcam not Working in 2.6 on M2 MacBook Air

ElDirector, 2024-04-25:
I have an M2 MacBook Air running Sonoma.  Version 2.3.1 was the first version that I installed and it worked just fine using both my iPhone or Logitech C922.  I just saw the 2.6 update and none of the cameras will load.  I also tried going backwards and the only version that works for me is 2.3.1.  All versions since do not load the webcams.  They will appear in the camera list but do not activate.

Maxou44, 2024-04-25:
Hello,

Thanks for your feedback, Have you an error message? Does the program crash? Or does the camera area shows a spinner?

Can you try on the web version here : https://app.eagle-animation.com/
And open the browser debug console to see if something happens ?

ElDirector, 2024-04-25:
Thank your for the quick reply.  I just tried the web version and it works, but only with the built-in webcam, not the external one I connected.  

I just launched the 2.6 executable via the Terminal and that actually works perfectly.  Here’s the output from that: 
Last login: Thu Apr 18 09:08:03 on console
/Applications/Eagle\ Animation\ 2.app/Contents/MacOS/Eagle\ Animation ; exit;
***@***.*** ~ % /Applications/Eagle\ Animation\ 2.app/Contents/MacOS/Eagle\ Animation ; exit;
💾 Eagle Animation files will be saved in the following folder: /Users/chris/Library/Application Support/EagleAnimation
📣 IPC GET_LAST_VERSION {}
📣 IPC GET_PROJECTS {}
📣 IPC SYNC {}
2024-04-25 12:39:49.504 Eagle Animation Helper (Plugin)[30426:2446397] WARNING: AVCaptureDeviceTypeExternal is deprecated for Continuity Cameras. Please use AVCaptureDeviceTypeContinuityCamera and add NSCameraUseContinuityCameraDeviceType to your Info.plist.
📣 IPC LIST_NATIVE_CAMERAS {}
📣 IPC GET_SETTINGS {}
📣 IPC APP_CAPABILITIES {}
📣 IPC GET_PROJECT { project_id: '868b82cb-354e-4f71-b863-b60924d5556a' }
📣 IPC LIST_NATIVE_CAMERAS {}
📣 IPC SAVE_PROJECT {
  project_id: '868b82cb-354e-4f71-b863-b60924d5556a',
  data: {
    project: {
      title: '',
      version: '2.3.1',
      creation: 1714062746,
      updated: 1714063697,
      deleted: false,
      scenes: [Array]
    },
    id: '868b82cb-354e-4f71-b863-b60924d5556a',
    preview: null
  }
}
📣 IPC GET_SETTINGS {}
📣 IPC SAVE_SETTINGS {
  settings: {
    CAMERA_ID: 'WEB-WEBCAM-6fe42f99bf41b737d64f137b098c67a5002b1ec1040f9970198253f5c089079f',
    FORCE_QUALITY: false,
    CAPTURE_FRAMES: 1,
    AVERAGING_ENABLED: false,
    AVERAGING_VALUE: 3,
    LANGUAGE: 'en',
    SHORT_PLAY: 20,
    RATIO_OPACITY: 1,
    GRID_OPACITY: 1,
    GRID_MODES: [ 'GRID', 'CENTER' ],
    GRID_LINES: 3,
    GRID_COLUMNS: 3,
    EVENT_KEY: '',
    SOUNDS: true
  }
}

But if I just try to launch the app like any other app, all my cameras appear in the camera list, but none of them work.  The black preview screen just has a blue ball animation.

Maxou44, 2024-04-25:
Can you confirm that if you launch the application from a terminal you can access all your cameras without any problem?

Did you install the application from the DMG or did you download the ZIP file? How do you run it once you've downloaded the file?

ElDirector, 2024-04-25:
If I run the executable from within the package contents, it launches the terminal and runs fine and all my cameras work.  If double click the application package from my Applications folder, the cameras are all listed, but none of them initialize.  

I installed this one from a DMG, but I did download the zip files for some previous ones to try that didn’t work.

Maxou44, 2024-04-26:
Can you try this version : https://github.com/brick-a-brack/eagle-animation/releases/tag/v2.6.1 ? 😀

ElDirector, 2024-04-26:
With this one, I get an error that the application is damaged and should be moved to the trash. It won’t open at all.

Maxou44, 2024-04-26:
Okey 😂

Maxou44, 2024-04-30:
Can you try v2.6.2 ? https://github.com/brick-a-brack/eagle-animation/releases/tag/v2.6.2

ElDirector, 2024-05-01:
It’s broken.  I can’t get the program to launch.

Maxou44, 2024-05-01:
Any errors from terminal?

Maxou44, 2024-05-02:
I've done some tests, and I think it's related to the fact that the application is built on an Intel Mac and not an ARM one. 

Rebuilding the application from an ARM Mac works without a problem, using the following steps :
- Instal NodeJS / NPM
- Clone sources
- Run `npm i --force`
- Run `npm run build:mac`
- Install the app built in the `binaries` folder

I'm trying to find a solution to fix this

Maxou44, 2024-05-05:
Can you have a look on this issue and check if it solves the issue on your side : https://github.com/electron-userland/electron-builder/issues/8191

ElDirector, 2024-05-05:
> Can you have a look on this issue and check if it solves the issue on your side : electron-userland/electron-builder#8191 <https://github.com/electron-userland/electron-builder/issues/8191>

Thank you for that, it fixed the issue.  I did have to rename to EagleAnimation.app instead of the space, but otherwise it’s perfect!

Thanks!!!

Maxou44, 2024-05-05:
Awesome, thanks for the feedback 🙏
I think we will write a guide for the setup 😊

ElDirector, 2024-05-05:
You’re welcome, and thank you for the help!  I’d be willing to record a short video tutorial on setting it up if you think that would help people?

Maxou44, 2024-05-06:
That would be great! 😀

ElDirector, 2024-05-06:
Here's my how-to video.  Let me know if you want anything changed
https://www.youtube.com/watch?v=ceGovao817g

Maxou44, 2024-05-07:
The video is perfect, thank you very much for taking the time to test and document it, I'm going to close the issue, I've added your video directly to the project README, it'll be easier. Feel free to open new issues if you have other ideas/needs. 😉

I'm glad to see that you like Eagle Animation and that you're using it in stop motion workshops, it's really nice to know that our software is being used ❤️

Version 2.7 is here, with a big performance boost 🚀

closed 2024-05-07 by Maxou44