brick-a-brack/eagle-animation #93: Camera Settings on Mac

ElDirector, 2024-06-12:
Hello!  I'm halfway through a kid's camp where we're using Eagle Animation for brickfilms.  We're using Windows computers and I noticed that when we plug in the Logitech cameras, there's a bunch of settings to control things like WB and Focus.  When I use EA on my MacBook, those options are not there.  I have the same Logitech C920 camera that we're using in class.

Maxou44, 2024-06-12:
Can you tell me which version of Eagle Animation you're using?

Perhaps it has something to do with the operating system? Camera control works under Windows (and normally under macOS, but I've never tested it).

Can you check on this page if you can control your camera settings on Chrome under macOS?
https://beaufortfrancois.github.io/sandbox/image-capture/playground.html

ElDirector, 2024-06-12:
I'm using 2.8.1 and there's no manual control settings.

Maxou44, 2024-06-13:
If you don't see camera settings on the webpage on Google Chrome, I think it's an issue related to the OS / Camera detection 😬

ElDirector, 2024-06-14:
I haven’t tried the online version, I’ll give it a shot. I’m currently using the dmg Sent from my iPhone

Maxou44, 2025-05-13:
Can you tell me if camera settings are working on the latest version?

ElDirector, 2025-05-13:
The camera is working, but I still don't have the settings.  Those are having to be controlled externally via 3rd party apps.

ElDirector, 2025-05-13:
I'll add, I'm using a Logitech C920.  I either have to use a third party camera control app to handle exposure and focus, or use the camera app from Logitech themselves.  I don't have in app controls like I do on Windows.

Maxou44, 2025-05-14:
Thanks for the updated feedback 😊

ElDirector, 2025-05-20:
Possibly a related issue: I'm using a third party camera control app to control focus and exposure since I can't do that directly from within the software.  Once I'm in Eagle Animation, if I leave it alone for too long, then the next time I take a photo, the exposure and focus is all messed up.  This has happened with both LogiTune and Camera Control third party apps, so I don't think it's related to them.

ElDirector, 2025-05-21:
Here's a video showing the two problems in action: 
[https://youtu.be/4EhFNiU-ms0](https://youtu.be/4EhFNiU-ms0)

Maxou44, 2025-05-21:
Thanks for your feedback. I think the first frame capture might produce a side effect due to camera initialization.

Did you have the same issues in previous versions?

Can you try on this page to see if you can control the camera settings: https://beaufortfrancois.github.io/sandbox/image-capture/playground.html

ElDirector, 2025-05-21:
I'm glad to help.  Yes, I've always had that issue on a Mac since I started using the software over a year ago.  Never had it on Windows.  

I tried the page in both Chrome and Safari, there are no camera settings other than select a camera, framerate and image size, though the image size sliders didn't affect anything for me.