brick-a-brack/eagle-animation #94: Save Settings

ElDirector, 2024-06-12:
Would it be possible to have an option to save settings?  Things like a default camera, checkboxs/switches (like auto focus), as well as export frame rates?  Even if they were just project based, rather than application based, it would help a lot with the kids I'm teaching.

Maxou44, 2024-06-12:
So save camera settings at project level and no longer at application level? 

What exactly do you use it for? Do you reopen a lot of old projects to continue them?

ElDirector, 2024-06-12:
Application level would be great.  In the class I'm teaching, kids are spending a few days on the same project and each time they open it we have to re-enable manual focus, and every time they export a shot, we have to manually set them to 24fps, even if we don't close the project.

Maxou44, 2024-06-14:
During our last LEGO exhibition, we also noticed that camera parameters were lost in some cases. I haven't figured out why, but it shouldn't be the case.

I split the issue for the FPS value reset, it's not related

Maxou44, 2025-05-08:
This bug was fixed with 5cc62a2743ff14f9101a25e42b26392124948234