brick-a-brack/eagle-animation #102: Mac app signing - Apple developer membership needed?

iainmerrick, 2024-06-18:
I installed the Mac app but was hit with a scary "this application is damaged, you should move it to the bin" message.

Looks like this is because it isn't signed, and the only workaround is to use `xattr` on the command line -- is that correct?

As Brick à Brack is a non-profit, it looks like you can get a waiver for the Apple Developer Program so you don't need to pay the $99 yearly fee: https://developer.apple.com/support/fee-waiver/

I'd love to help out with Mac feature development, if you're able to get the developer membership sorted out. In particular I'm interested in adding support for Canon cameras.

Maxou44, 2024-06-19:
Hello,

You're totally right, Mac/Windows builds are not signed yet, I'll find out how to get an Apple developper account and start the process as a non-profit organization, thanks for the information!  🚀

I would be happy to have some help with camera issues on Apple devices, since I don't have the equipment to test and debug them. 😅

We use this module to control Canon cameras: https://github.com/brick-a-brack/napi-canon-cameras but I only built it for Windows, the first step would be to build it on Mac and update the pipeline :) #5 

It might also be worthwhile for the Linux and Mac versions to take a look at https://github.com/tsedio/ts-gphoto2-driver to fix camera settings issues mentionned in #93 and add various cameras support.

In the future it could also be awesome to support the WebAssembly version of gPhoto2 to support USB cameras directly on the web version on Mac/Linux #69

Don't hesitate if you have any questions, I'm also available on Discord as @Maxou44 if needed 😊

Maxou44, 2024-06-22:
I've sent a request for the developer account, I'm waiting for Apple to get back to me.