brick-a-brack/eagle-animation #109: Provide a `rpm`

stefanfoulis, 2024-10-11:
Provide an `rpm` as part of the release.
(In my case for Fedora)

Maybe as a intermediate solution: provide documentation on how to run eagle-animation from the generic linux release.

Maxou44, 2024-10-12:
Thanks for your feedback, I will try to bundle it as rpm file in the future 👌

You can manually build it using NodeJS using the following commands :
- npm i --force
- npm run build:linux 

The web version is also available in https://app.eagle-animation.com

stefanfoulis, 2024-10-19:
On second thought about providing an `rpm` release.... Maybe it would make more sense to provide a `flatpak` or `appimage` :thinking: . Then we'd reach more different linux distros with one additional package type.

Maxou44, 2024-10-27:
The next version (2.9.x) will be bundled as rpm, AppImage and flatpak. The app binary have been renamed from "@brick-a-brickeagle-animation" to "eagle-animation".

closed 2024-10-27 by Maxou44

stefanfoulis, 2024-12-05:
I briefly tried out the new AppImage in 2.9.0 today (on Fedora 41 in Gnome) and everything seems to work :+1:

Maxou44, 2024-12-05:
Awesome! Thanks for the feedback! 🙂👌