brick-a-brack/eagle-animation #116: Web export difficulty on Chromebook. Recovery options?

denneyliptak, 2024-12-08:
Greetings!

My daughter has some work accrued using the web version. She's using a chromebook.

The web version's capture view still loads the thumbnails and she's able to continue with the project. 

However, export (either to movie or individual images) seems to fail. On one project, the "export successful" screen comes up but then the browser window resets to the "Aww snap" out of memory message.

Is there any way I can manually get these files off, perhaps using the filesystem? Or if they're in a blob of some type, can I move that off to another machine with more oomph?

Thank you much,
--Denney

Maxou44, 2024-12-08:
Good evening,

I'm sorry to see that exporting to the web version doesn't work and that you're having a problem.

➡️ Have you tried relaunching the application and exporting the images in original quality (this should not involve image compression and should be faster)?
➡️ Can you check in the browser console if there is a specific warning or error ?
➡️ How many photos have you taken in the project?

For the web version, frames are store in your browser IndexedDB, in “BlobFramesDatabase” and "ProjectDatabase", when you export the frames, it tries to create a ZIP file, when you try to export a video, it will use ffmpeg.wasm to export the video file.

If there's still a memory problem in the browser, I'll add a feature to allow you to download the photos without creating a ZIP, which should allow you to recover your photos without any problem.

denneyliptak, 2024-12-09:
Hello! Thank you for the reply :)

- True; initially we tried exporting the video and then switched to the individual frames without any resizing.
- For some reason "Developer Tools" is gray on her chromebook...if I could get that enabled, I'd be happy to help troubleshoot the console messaging.
- She's got a few projects :-\  I think maybe a little over 1000 images total if we summed them all. I believe they are 1080p captures from a webcam.

I installed 'IndexedDB Exporter' and was indeed able to see the "BlobFramesDatabase" but the browser eventually crashes trying to export the 'BlobFramesDatabase'. (no message even...just disappears/closes down)

I wasn't sure how this 'magic' happened (you've helped me learn there's a DB behind the scenes, saving the images locally) and I advised my daughter to export her work each day...but alas.

I very much appreciate your offer for the custom image export; if that process could happen image-by-image maybe her poor ol' chromebook wouldn't run out of memory. I'm happy to run ffmpeg on the resulting stack of files, no problem there.

Maxou44, 2024-12-12:
I just released the 2.10.0, it will use the Chromium **Filesystem API** for frames exportation instead of a ZIP generation.

Let me know if this solves the problem and allows you to export your frames without issues 🙂

denneyliptak, 2024-12-13:
I've verified 2.10 web-view loads on my main PC and did a frame (original) export. If I select a folder, it will write individual files out as you explained. Cool! If I try again, only I close the window instead of choosing a folder, it appears to fall back to generating the zip, which is also good.

However, on her chromebook, it never pulled up a dialog to choose a folder. It seemed to begin processing (the progress even got to a 'complete' message) but eventually crashed...perhaps it chose the .zip route.

I think you've made a nice change and it works well on my PC. I just need to figure out how to purge as much as I can from RAM on the chromebook. It's only got 4GB and I think only around 500MB is 'free'. It's frustrating how locked down it is. (no actual file browser, no developer tools in the browser)

Maxou44, 2024-12-22:
Hello,

Thank you for your last message, and sorry for the delay in my reply.

I've just released a new version 2.10.1, which lets you choose whether you want a ZIP file or not. If ZIP is disabled and Chrome FS API not available, the application triggers a download per photo.

Let me know if it's good 😉

Maxou44, 2025-01-18:
Any news @denneyliptak  ? 😊

denneyliptak, 2025-01-18:
Thank you for checking back, and for the help! Sadly, her Chromebook crashes trying to load the site now. I think the HW is just too old and I can't lighten the memory pressure.