brick-a-brack/eagle-animation #122: UI improvement - Onion blending slider not obvious

Jlectric, 2025-03-11:
As a new user, the onion skin UI option was really unintuitive to find.
It is only labelled with a tooltip, adding some text above it or an icon would be great!
[image]

Nice to have:
_I turned `Difference` mode on first and then tried moving the slider - with difference mode on, the slider doesn't do anything._
It would be nice if the UI indicated this too. (Maybe grey out the onion skin slider when difference mode is on?)

Maxou44, 2025-03-12:
Thanks for your issue, I'll try to improve that soon :)

Maxou44, 2025-05-08:
In the next update, when the difference mode is enabled, the slider will be disabled :)
[image]