brick-a-brack/eagle-animation #126: 2.11.0 Crashing on Mac

ElDirector, 2025-05-12:
I'm getting ready for another round of kid's college, so I downloaded and installed 2.11.0 on my M2 MacBook Air.  

After using the xattr -c /Applications/eagle-animation.app fix in Terminal, the app opens, and I can access settings, but if I try to load or create a project, I get an error that something wrong happened and the app needs to restart.  I am unable to load or start a new project at this time.  

I tried to run the program through Terminal and get the following error:

2025-05-12 12:58:15.665 Eagle Animation Helper (Plugin)[23286:1668143] WARNING: Add NSCameraUseContinuityCameraDeviceType to your Info.plist to use AVCaptureDeviceTypeContinuityCamera.
2025-05-12 12:58:16.300 Eagle Animation Helper (Plugin)[23286:1668143] WARNING: AVCaptureDeviceTypeExternal is deprecated for Continuity Cameras. Please use AVCaptureDeviceTypeContinuityCamera and add NSCameraUseContinuityCameraDeviceType to your Info.plist.

I'm unable to update the Info.plist

Maxou44, 2025-05-13:
Thanks for the report, I'll check that later today :)

Maxou44, 2025-05-13:
Can you try with v2.11.1 ? 😊

ElDirector, 2025-05-13:
Sure, is there a link to a dmg?  I can't find that one

Maxou44, 2025-05-13:
Available here :https://github.com/brick-a-brack/eagle-animation/releases/tag/v2.11.1

ElDirector, 2025-05-13:
That one seems to work!  I'll test it out over the next day or so and see how it performs :). Thanks!

Maxou44, 2025-05-13:
Thanks the feedback, I'll close the issue :)

closed 2025-05-13 by Maxou44