brick-a-brack/eagle-animation #185: Web export difficulty on Linux

kattelk, 2025-07-27:
I have tried to export a short film using version 2.11.2 on two different Linux distributions (Manjaro and Fedora).  I export the film as a video, choose the file's location. Then, during the export, the progress percentage increases, but never reaches 100, and I get the message "The export is finished, you can close this page."  The file however does not exist.

Exporting animation frames works as expected.

Can you help me with this?

Maxou44, 2025-07-27:
Can you try to start Eagle Animation from a terminal and send us the logs available in the terminal ?

kattelk, 2025-07-27:
```
💾 Eagle Animation files will be saved in the following folder: /home/user/.local/share/EagleAnimation
[45211:0727/225759.736503:ERROR:ui/gl/gl_surface_presentation_helper.cc:260] GetVSyncParametersIfAvailable() failed for 1 times!
[45169:0727/225803.677514:ERROR:storage/browser/quota/quota_database.cc:1031] Could not open the quota database, resetting.
[45211:0727/225803.813888:ERROR:ui/gl/gl_surface_presentation_helper.cc:260] GetVSyncParametersIfAvailable() failed for 2 times!
[45169:0727/225804.713709:ERROR:content/browser/indexed_db/instance/leveldb/backing_store.cc:226] Failed to open LevelDB database from /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb,IO error: /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb/LOCK: No further details. (ChromeMethodBFE: 15::LockFile::1)
[45169:0727/225805.716703:ERROR:content/browser/indexed_db/instance/leveldb/backing_store.cc:226] Failed to open LevelDB database from /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb,IO error: /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb/LOCK: No further details. (ChromeMethodBFE: 15::LockFile::1)
[45169:0727/225806.718758:ERROR:content/browser/indexed_db/instance/leveldb/backing_store.cc:226] Failed to open LevelDB database from /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb,IO error: /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb/LOCK: No further details. (ChromeMethodBFE: 15::LockFile::1)
[45211:0727/225806.948911:ERROR:ui/gl/gl_surface_presentation_helper.cc:260] GetVSyncParametersIfAvailable() failed for 3 times!
[45169:0727/225807.728277:ERROR:content/browser/indexed_db/instance/leveldb/backing_store.cc:226] Failed to open LevelDB database from /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb,IO error: /home/katt/.config/eagle-animation/IndexedDB/file__0.indexeddb.leveldb/LOCK: No further details. (ChromeMethodBFE: 15::LockFile::1)

```
and then the last line repeats.

I can only see animation frames in the .local/share folder, no videos.

Maxou44, 2025-07-27:
Did you launched a video export, normally there is some logs about it 😏

Maxou44, 2025-07-28:
I’ve never seen this error 234 with FFmpeg.
I don’t see a file extension in the command line — can you try exporting as "video.mp4" instead of just "video"?

kattelk, 2025-07-29:
That worked! Thank you!!

Shouldn't the default name for the video be "video.mp4", and not just "video"?  Or the file format selector include the supported file formats?

commit c20bbff4 (2025-08-10): fix: Add extension to filename if missing, fixes #185